dj1ch/minigotchi-ESP32 #27: Minigotchi crashing with Config::advertise = true;

Bender1983, 2024-06-14:
When setting Config::advertise = true; in config then minigotchi crashes

platform: Cardputer M5 (ESP32-s3fn8)

Here is the output of the console with the backtrace:
 `
 (^-^) Hi, I'm Minigotchi, your pwnagotchi's best friend!
 
('-') You can edit my configuration parameters in config.cpp!
 
(>-<) Starting now...
 
################################################
#                BOOTUP PROCESS                #
################################################
 
('-') Adding XYZ to the whitelist
('-') Adding 123 to the whitelist
('-') Adding 456 to the whitelist
 
(-.-) Initializing on channel 1
 
(X-X) Channel initialization failed, try again?
 
((0-o) Scanning for Pwnagotchi.
(o-0) Scanning for Pwnagotchi..
(0-o) Scanning for Pwnagotchi...
 
(0-o) Scanning for Pwnagotchi.
(o-0) Scanning for Pwnagotchi..
(0-o) Scanning for Pwnagotchi...
 
(0-o) Scanning for Pwnagotchi.
(o-0) Scanning for Pwnagotchi..
(0-o) Scanning for Pwnagotchi...
 
(;-;) No Pwnagotchi found
 
(>-<) Starting advertisment...
 
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x4205a3de  PS      : 0x00060c30  A0      : 0x8205a558  A1      : 0x3fcebb10  
A2      : 0x00000000  A3      : 0x3fcec8d8  A4      : 0x3fcef91c  A5      : 0x00000000  
A6      : 0x3fcef940  A7      : 0x00000000  A8      : 0x8200739a  A9      : 0x3fcebaf0  
A10     : 0x00000000  A11     : 0x3fcec3e0  A12     : 0x000004e6  A13     : 0x3fcecdbb  
A14     : 0x3fcecdbb  A15     : 0x3fcecb6c  SAR     : 0x0000001d  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000054  LBEG    : 0x400556d5  LEND    : 0x400556e5  LCOUNT  : 0xffffffff  

Backtrace: 0x4205a3db:0x3fcebb10 0x4205a555:0x3fcebb50 0x4200a219:0x3fcebb80 0x4200a352:0x3fcebba0 0x4200acb3:0x3fcebc10 0x4200a541:0x3fcebc30 0x420175a4:0x3fcebc50

ELF file SHA256: 9cda293c6cded480
`

After a lot of try and error to identify the error, I think the error is somwhere around the Frame::send() part in void Frame::advertise() at frame.cpp but was unable to identify it better.

dj1ch, 2024-06-14:
I know the issue is being worked on, but I haven't found much off of exception decoders.

dj1ch, 2024-06-15:
nah it's all good. I'm in the progress of fixing it but I haven't really close to fixing the exception

ATOMNFT, 2024-06-16:
I think this is one problematic area. Instead of appending each byte from Frame::beaconFrame, it should be appending data from Frame::payload.

ATOMNFT, 2024-06-16:
> I think this is one problematic area. Instead of appending each byte from Frame::beaconFrame, it should be appending data from Frame::payload.
> The pack() function constructs the entire beacon frame by first initializing it (init()), appending ESSID data (essid()), and then chunking the data into packets using chunkSize. There's a logical error in how the payload is being copied:

I could be totally wrong though...

dj1ch, 2024-06-16:
> I think this is one problematic area. Instead of appending each byte from Frame::beaconFrame, it should be appending data from Frame::payload. The pack() function constructs the entire beacon frame [...]

`Frame::payload` doesn't exist. there's `Frame::payloadSize` which is used to measure the length of the essid + header without the chunking. 

```cpp
/**
 * frame.cpp
 */

void Frame::pack() {
  // clear frame before constructing
  Frame::beaconFrame.clear();

  // add the header and essid
  init();
  essid();

  // payload size
  Frame::payloadSize = Frame::beaconFrame.size();

  // store for later
  std::vector<uint8_t> originalBeaconFrame = Frame::beaconFrame;

  for (size_t i = 0; i < payloadSize; i += Frame::chunkSize) {
    Frame::beaconFrame.push_back(Frame::IDWhisperPayload);

    size_t chunkEnd = std::min(i + Frame::chunkSize, Frame::payloadSize);
    Frame::beaconFrame.push_back(chunkEnd - i);

    for (size_t j = i; j < chunkEnd; ++j) {
      Frame::beaconFrame.push_back(originalBeaconFrame[j]);
    }
  }

  // save some memory
  originalBeaconFrame.clear();
  originalBeaconFrame.shrink_to_fit();

  // update size since we changed the frame
  Frame::frameSize = Frame::beaconFrame.size();

}
```

When constructing the frame there are a couple of steps that need to be done

1. Header (appended to `Frame::beaconFrame`)
2. JSON Data (appended to `Frame::beaconFrame`)
3. For every 255 bytes we will need to chunk the payload and throw in something called `Frame::IDWhisperPayload` to indicate this is from a Pwnagotchi

Something I've noticed with ESP32 development is that the compiler is a lot more strict when it comes to declaring objects, function signatures, etc. I'll probably need to adjust accordingly. I'll read up more on `pack.go` from pwngrid if needed

dj1ch, 2024-06-20:
@ATOMNFT and I just recently solved the issue, if you want to try compiling from the development branch for the ESP32.

dj1ch, 2024-06-21:
Going to close this as advertising works now.

closed 2024-06-21 by dj1ch