dj1ch/minigotchi-ESP32 #41: Got wifi unsupport

hamabao413, 2024-06-28:
**Describe the bug**
I Got wifi : unsupport fram type : 0c0
I Got wifi : unsupport fram type : 0a0
and 
Screen show Both packets failed to send

Dont know what I did err

dj1ch, 2024-06-28:
> **Describe the bug** I Got wifi : unsupport fram type : 0c0 I Got wifi : unsupport fram type : 0a0 and Screen show Both packets failed to send
> 
> Dont know what I did err

esp-idf will block deauthentication frames by default. in this case you will need to apply the fixes stated in the install guide in [this section](https://github.com/dj1ch/minigotchi-ESP32/blob/main/INSTALL.md#step-2-building-and-flashing)

dj1ch, 2024-06-28:
let me know if this fix works, I have applied this fix to my ESP32 already so it should.

hamabao413, 2024-06-29:
> let me know if this fix works, I have applied this fix to my ESP32 already so it should.

I'm not sure if the format is incorrect when I editing platform.txt. Is there any demonstration?
I did:    build.extra_flags.esp32 -w
            build.extra_flags.esp32= -w
 both r err 

Thank you for help

dj1ch, 2024-06-29:
> I'm not sure the format is incorrect when I editing platform.txt. Is there any demonstration?
> I did:
> build.extra_flags.esp32=-w -DARDUINO_USB_CDC_ON_BOOT=0
> compiler.c.elf.libs.esp32=-zmuldefs -lesp_ringbuf -lefuse -lesp_ipc -ldriver -lesp_pm -lmbedtls -lapp_update -lbootloader_support -lspi_flash -

As of now, we don't have any video guides. But from the look of it this is correct if that was the only modification you made. You're supposed to use esp32 version 2.0.10 not 2.0.11 because this can lead to unknown issues, which could be the cause of this problem. The fix is meant to be applied to that version only.

ATOMNFT, 2024-06-29:
> > let me know if this fix works, I have applied this fix to my ESP32 already so it should.
> 
> I'm not sure the format is incorrect when I editing platform.txt. Is there any demonstration? I did:
> 
> build.extra_flags.esp32=-w -DARDUINO_USB_CDC_ON_BOOT=0
> 
> compiler.c.elf.libs.esp32=-zmuldefs -lesp_ringbuf -lefuse -lesp_ipc -ldriver -lesp_pm -lmbedtls -lapp_update -lbootloader_support -lspi_flash -
> 
> Same, I Got wifi : unsupport fram type : 0c0 　　　I Got wifi : unsupport fram type : 0a0
> 
> is that because i am using ESP32 version 2.0.11 not 2.0.10?
> 
> Also in INSTALL.md the II For compiler.c.elf.libs.esp32, compiler.c.elf.libs.esp32s2, compiler.c.elf.libs.esp32s3, compiler.c.elf.libs.esp32c3, add -zmuldefs to their compile settings III For compiler.c.elf.libs.esp32, compiler.c.elf.libs.esp32s2, compiler.c.elf.libs.esp32s3, compiler.c.elf.libs.esp32c3, add -zmuldefs to their compile settings are the same step?
> 
> **ESP8266 works for me and having fun**
> 
> Thank you for help and your great work!!

Yes, go to version 2.0.10
It is mentioned in the marauder tutorial (https://github.com/justcallmekoko/ESP32Marauder/wiki/arduino-ide-setup) that other versions will not work.

dj1ch, 2024-06-29:
> Yes, go to version 2.0.10 It is mentioned in the marauder tutorial (https://github.com/justcallmekoko/ESP32Marauder/wiki/arduino-ide-setup) that other versions will not work.

I figured that was the issue anyway. I'll close this as soon as their ESP32 works correctly.

hamabao413, 2024-06-30:
> I figured that was the issue anyway. I'll close this as soon as their ESP32 works correctly.

OK thank you!!  I will try to install again.

hamabao413, 2024-07-01:
> OK thank you!! I will try to install again.

i installed ESP32 by Espressif System 2.0.10 but is same just got wifi : unsupport fram type : 0c0

dj1ch, 2024-07-02:
did you make the changes to platform.txt on `2.0.10`?

hamabao413, 2024-07-02:
> did you make the changes to platform.txt on `2.0.10`?

Yes, I did and checked many times, Also tried Lilygo ESP32-S without display.

dj1ch, 2024-07-02:
Are you compiling from the development branch or are you using a release? the latest ESP32 release has issues with deauthing(a couple months old)

hamabao413, 2024-07-03:
I am using latest ESP32 release..... I thought that is NEW, I will try download files later. Thank you!

hamabao413, 2024-07-03:
> > I am using latest ESP32 release..... I thought that is NEW, I will try download files later. Thank you!
> 
> no problem. at the moment I'm a little busy so most of my progress is in the development branch. once I can fix everything when I get the chance I can probably make a new release.

Confirmed that it works!!!!  Good Job ! I love it!

dj1ch, 2024-07-03:
going to close the issue now that the problem is resolved.