dj1ch/minigotchi-ESP32 #47: Compilation error: ESP8266WiFi.h: No such file or directory

Lordex-x, 2024-07-04:
At the time of compiling, it gives an error in .bin, warning that the file 'ESP8266WiFi.h' was not found.

I followed the installation step by step for the cardputer, with the minigotchi open in the Arduino IDE, I clicked on 'sketch' and 'export compiled binary'.

I expected a new executable .bin file to be generated. If I did something wrong, can you teach me?"

Let me know if there's anything else you'd like to know!

[image]

matrix224, 2024-07-04:
I think you downloaded the wrong repo.
There's two:

https://github.com/dj1ch/minigotchi
and 
https://github.com/dj1ch/minigotchi-ESP32

The first one is for ESP8266 boards, and the second one (the repo this issue is posted on) is for ESP32. It looks like you downloaded the ESP8266 one, but you want the ESP32 one.

Try downloading the source from this repo and trying with that instead

Lordex-x, 2024-07-04:
> I think you downloaded the wrong repo.
> There's two:
>
> https://github.com/dj1ch/minigotchi
> and
> https://github.com/dj1ch/minigotchi-ESP32
>
> The first one is for ESP8266 boards, and the second one (the [...]

Thank you very much for the help, I'm already fixing it, I'll probably have more issues soon

dj1ch, 2024-07-08:
I'll close this issue as this one is solved. Thank you @matrix224

closed 2024-07-08 by dj1ch