dj1ch/minigotchi-ESP32 #48: xtensa-esp32s3-elf-g++: error: zmuldefs: No such file or directory  exit status 1  Compilation error: exit status 1

Lordex-x, 2024-07-05:
During compilation, an error occurs: file zmuldefs.h not found.

I followed the installation steps for the cardputer to generate a binary compiled by the IDE.

I expected to create a .bin file.

This is the second step in the installation guide.
[image]

ATOMNFT, 2024-07-07:
> During compilation, an error occurs: file zmuldefs.h not found.
> 
> I followed the installation steps for the cardputer to generate a binary compiled by the IDE.
> 
> I expected to create a .bin file.
> 
> This is the second step in the installation guide. ![imagem_2024-07-05_102643672](https://private-user-images.githubusercontent.com/174689527/346153365-b70f169d-7bbe-4392-8128-314b840b347a.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjAzMTQ5NTksIm5iZiI6MTcyMDMxNDY1OSwicGF0aCI6Ii8xNzQ2ODk1MjcvMzQ2MTUzMzY1LWI3MGYxNjlkLTdiYmUtNDM5Mi04MTI4LTMxNGI4NDBiMzQ3YS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNzA3JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDcwN1QwMTEwNTlaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT01Y2ViNjk4ZmQ5Mzk0YWMxZWRlZTRmMDBhZTY0ZWRmNDk5YjdmYmQ4ZWEyYTM3NTBhNWYxNmMwYzUxYThlMTFhJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.H_TmIC8dDIF4mHZ-QWQ1dVk3b1wHkKJHmY-zTSwmXD4)

Have you made the edits to platform.txt file?

Lordex-x, 2024-07-07:
I edited
[image]

ATOMNFT, 2024-07-07:
> I edited
> [image]
> 
> 

Before or after the error you posted? Also did you add the -w to the correct locations in platform.txt?

Lordex-x, 2024-07-07:
, I edited before and also edited the "-w" parts."

dj1ch, 2024-07-08:
Seems to compile fine on my end. From the look of it you're compiling from the main or development branch. Which one are you using?

dj1ch, 2024-07-08:
Just merged `main` and `development` in case you want the most recent changes.

Lordex-x, 2024-07-09:
> Parece compilar bem do meu lado. Pelo que parece, você está compilando a partir do branch principal ou de desenvolvimento. Qual você está usando?

main

Lordex-x, 2024-07-09:
I deleted everything and downloaded it again, and the error didn't repeat itself, so it was probably some typing mistake on my part in some unknown part of the code. But I'm already having difficulties with the 'ieee80211' part.

Lordex-x, 2024-07-09:
I managed to compile successfully. Where can I find the binary file?

dj1ch, 2024-07-09:
> I managed to compile successfully. Where can I find the binary file?

should be in the directory `minigotchi-ESP32/build/<name of your ESP32>`.

dj1ch, 2024-07-09:
> I deleted everything and downloaded it again, and the error didn't repeat itself, so it was probably some typing mistake on my part in some unknown part of the code. But I'm already having difficulties with the 'ieee80211' part.

wdym by the ieee80211 part? like is deauthing not working or something? most of the networking part is handled by the minigotchi.

Lordex-x, 2024-07-09:
> > Apaguei tudo e baixei novamente, e o erro não se repetiu, então provavelmente foi algum erro de digitação da minha parte em alguma parte desconhecida do código.  Mas já estou tendo dificuldades com a parte 'ieee80211'.
> 
> wdym pela parte ieee80211? tipo, a deauthing não está funcionando ou algo assim? a maior parte da parte de rede é gerenciada pelo minigotchi.

Redundancy of files in the library.

dj1ch, 2024-07-09:
> Redundancy of files in the library.

I'm not sure what you mean, can you show me the output from the Minigotchi or the Arduino compiler which has the error?

Lordex-x, 2024-07-09:
I had similar files from "M5Stack" in the library that were conflicting during compilation with those from "ESP32".

dj1ch, 2024-07-10:
Ah okay. I'm assuming it works fine now?

ATOMNFT, 2024-07-10:
Were you getting "Multiple Definition of ieee80211_raw_frame_sanity_check" ?

ATOMNFT, 2024-07-10:
> I deleted everything and downloaded it again, and the error didn't repeat itself, so it was probably some typing mistake on my part in some unknown part of the code. But I'm already having difficulties with the 'ieee80211' part.

Were you getting "Multiple Definition of ieee80211_raw_frame_sanity_check" ?

dj1ch, 2024-07-18:
I am going to close this issue as it seems to be solved.