dj1ch/minigotchi-ESP32 #51: New update cant install

hamabao413, 2024-07-09:
I download ur new code and try to install but i got err.

c:/users/tommy/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32s3-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: C:\Users\tOmmy\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.10/tools/sdk/esp32s3/lib\libnet80211.a(ieee80211_output.o): in function `ieee80211_raw_frame_sanity_check':
(.text.ieee80211_raw_frame_sanity_check+0x5c): multiple definition of `ieee80211_raw_frame_sanity_check'; C:\Users\tOmmy\AppData\Local\Temp\arduino\sketches\1185B356109DB8315BD3A46138052B5C\sketch\deauth.cpp.o:D:\tOmmy\Desktop\minigotch\new ESP32\minigotchi-ESP32/deauth.cpp:36: first defined here
collect2.exe: error: ld returned 1 exit status

Using library Adafruit GFX Library at version 1.11.9 in folder: D:\tOmmy\Documents\Arduino\libraries\Adafruit_GFX_Library 
Using library Adafruit BusIO at version 1.16.1 in folder: D:\tOmmy\Documents\Arduino\libraries\Adafruit_BusIO 
Using library Wire at version 2.0.0 in folder: C:\Users\tOmmy\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.10\libraries\Wire 
Using library SPI at version 2.0.0 in folder: C:\Users\tOmmy\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.10\libraries\SPI 
Using library Adafruit SSD1305 at version 2.2.3 in folder: D:\tOmmy\Documents\Arduino\libraries\Adafruit_SSD1305 
Using library Adafruit SSD1306 at version 2.5.10 in folder: D:\tOmmy\Documents\Arduino\libraries\Adafruit_SSD1306 
Using library TFT_eSPI at version 2.5.43 in folder: D:\tOmmy\Documents\Arduino\libraries\TFT_eSPI 
Using library FS at version 2.0.0 in folder: C:\Users\tOmmy\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.10\libraries\FS 
Using library SPIFFS at version 2.0.0 in folder: C:\Users\tOmmy\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.10\libraries\SPIFFS 
Using library U8g2 at version 2.34.22 in folder: D:\tOmmy\Documents\Arduino\libraries\U8g2 
Using library WiFi at version 2.0.0 in folder: C:\Users\tOmmy\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.10\libraries\WiFi 
Using library ArduinoJson at version 7.1.0 in folder: D:\tOmmy\Documents\Arduino\libraries\ArduinoJson

dj1ch, 2024-07-09:
Looks like the error is in your `platform.txt`. Is it correctly modified? This error should occur when `platform.txt` isn't correct.

hamabao413, 2024-07-10:
> Looks like the error is in your `platform.txt`. Is it correctly modified? This error should occur when `platform.txt` isn't correct.

Yes, U r right,  dont know when changed.....but I having hard time to make T-display S3 work, Just become black, not just in your code.... Thank you for your help

dj1ch, 2024-07-10:
> > Looks like the error is in your `platform.txt`. Is it correctly modified? This error should occur when `platform.txt` isn't correct.
> 
> Yes, U r right, dont know when changed.....but I having hard time to make T-display S3 work, Just become black, not just in your code.... Thank you for your help

I'm using one too. In this case for `bool Config::display`, set that to `true`, then for `std::string Config::screen` set that to `"T_DISPLAY_S3"`

hamabao413, 2024-07-10:
> > > Looks like the error is in your `platform.txt`. Is it correctly modified? This error should occur when `platform.txt` isn't correct.
> > 
> > 
> > Yes, U r right, dont know when changed.....but I having hard time to make T-display S3 work, Just become black, not just in your code.... Thank you for your help
> 
> I'm using one too. In this case for `bool Config::display`, set that to `true`, then for `std::string Config::screen` set that to `"T_DISPLAY_S3"`. I'll be sure to edit and revise the current documentation.

The settings are correct. I just replaced all the Lib files and now fully usable. Thnak you so much!! Hope to see your work grow!!

dj1ch, 2024-07-10:
No problem. I'll close this issue now.

closed 2024-07-10 by dj1ch