dj1ch/minigotchi-ESP32 #61: Compile error

luizmucman, 2024-07-23:
In compile i got this error

c:/users/lhrc/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32s2-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32s2-elf/8.4.0/../../../../xtensa-esp32s2-elf/bin/ld.exe: C:\Users\lhrc\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.10/tools/sdk/esp32s2/lib\libnet80211.a(ieee80211_output.o): in function `ieee80211_raw_frame_sanity_check':
(.text.ieee80211_raw_frame_sanity_check+0x5c): multiple definition of `ieee80211_raw_frame_sanity_check'; C:\Users\lhrc\AppData\Local\Temp\arduino\sketches\9ED0C7C9D467676F252000F410EA4932\sketch\deauth.cpp.o:C:\Users\lhrc\Downloads\minigotchi-ESP32-3.3.2-beta\minigotchi-ESP32-3.3.2-beta\minigotchi-ESP32/deauth.cpp:36: first defined here
collect2.exe: error: ld returned 1 exit status

exit status 1

Compilation error: exit status 1

Eladji, 2024-07-23:
That a normal issue you need to add some compile parameters 
At

** ~/.arduino15/packages/esp32/hardware/esp32/{version}/platform.txt**

then
-zmuldefs to compiler.c.elf.libs.esp32
-w to build.extra_flags.esp32

this solution is from this [repo](https://github.com/tesa-klebeband/ESP32-Deauther-ArduinoIDE)

dj1ch, 2024-07-23:
There's a step in the install guide which covers this 

https://github.com/dj1ch/minigotchi-ESP32/blob/main/INSTALL.md#step-2-building-and-flashing

ATOMNFT, 2024-07-24:
Here is a vid and still image showing the edits needed.

https://github.com/user-attachments/assets/ec6e0239-db02-4db1-9857-00c9f3d851e4

[image]

dj1ch, 2024-07-29:
Seems like it's been a good week or so. Going to close this issue since there's no activity.

closed 2024-07-29 by dj1ch