dj1ch/minigotchi-ESP32 #66: Error with Wifi.h and my_ieee80211_raw_frame_sanity_check

AbeNaws, 2024-08-06:
The code will not compile (I assume after the formatting update)

I was able to trace the issue to frame.h with line 30:
`#include <Wifi.h>`
When it needs to be 
`#include <WiFi.h>`

After that I got an error in deauth.cpp that was fixable by renaming 
`ieee80211_raw_frame_sanity_check` on line 35 to 
`my_ieee80211_raw_frame_sanity_check`

dj1ch, 2024-08-07:
> The code will not compile (I assume after the formatting update)
> I was able to trace the issue to frame.h with line 30:
#include <Wifi.h>
When it needs to be
#include <WiFi.h>

Looks like the reformatter probably did that. I'll fix that in a sec. 

> After that I got an error in deauth.cpp that was fixable by renaming
ieee80211_raw_frame_sanity_check on line 35 to
my_ieee80211_raw_frame_sanity_check

https://github.com/dj1ch/minigotchi-ESP32/blob/main/INSTALL.md#step-2-building-and-flashing

This section of the install guide covers this. You will need to apply this fix as it allows you to do deauthing.

AbeNaws, 2024-08-07:
Sorry if this is obvious, but what part of the instructions applies to that?

dj1ch, 2024-08-07:
The part where you need to edit your `platform.txt`, it modifies esp-idf so that the function in `deauth.cpp` works without running into that error

dj1ch, 2024-08-07:
> Looks like the reformatter probably did that. I'll fix that in a sec.
>
>
> https://github.com/dj1ch/minigotchi-ESP32/blob/main/INSTALL.md#step-2-building-and-flashing
>
> This section of the install guide [...]

Just noticed that this change probably happened a really long time ago with `WiFi.h`. It is unbeknownst to me on why my IDE(both vscode and Arduino IDE) have ignored it.

dj1ch, 2024-08-08:
Going to close this issue due to no activity.

closed 2024-08-08 by dj1ch

codingneko, 2024-11-12:
Not sure this is related, but this error might help shed some light, also having problems to compile. Sorry for reviving a dead issue.

/home/cody/.arduino15/packages/esp32/tools/esp-xs3/2302/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld: /home/cody/.arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-33fbade6/esp32s3/lib/libnet80211.a(ieee80211_output.o): in function `ieee80211_raw_frame_sanity_check':
(.text.ieee80211_raw_frame_sanity_check+0x5c): multiple definition of `ieee80211_raw_frame_sanity_check'; /tmp/arduino/sketches/047A3FEBF3D9B4EC2B219CBE10825E4C/sketch/deauth.cpp.o:/home/cody/Downloads/minigotchi-ESP32/minigotchi-ESP32/deauth.cpp:36: first defined here
collect2: error: ld returned 1 exit status

My setup:
- Arduino IDE
- Heltec WiFi Kit 32(V3)
- Compiling on Arduino IDE 2.3.3 running on Linux 6.8.0-48-generic

dj1ch, 2024-11-12:
> Not sure this is related, but this error might help shed some light, also having problems to compile. Sorry for reviving a dead issue. [...]

Yeah, this is the same issue. It can be solved in [this](https://github.com/dj1ch/minigotchi-ESP32/blob/main/INSTALL.md#step-2-building-and-flashing) section of the install guide. It needs to be done in order to make sure deauthing works properly. Thanks for asking.