dj1ch/minigotchi-ESP32 #80: Compile Error

ririinsta, 2024-08-19:
```
In file included from C:\Users\Admin\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.1-b6b4727c58\esp32c6/include/esp_wifi/include/esp_private/esp_wifi_private.h:14,
                 from C:\Users\Admin\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.1-b6b4727c58\esp32c6/include/esp_wifi/include/esp_wifi.h:57,
                 from C:\Users\Admin\Downloads\minigotchi-ESP32\deauth.h:32,
                 from C:\Users\Admin\Downloads\minigotchi-ESP32\parasite.h:28,
                 from C:\Users\Admin\Downloads\minigotchi-ESP32\channel.h:29,
                 from C:\Users\Admin\Downloads\minigotchi-ESP32\minigotchi.h:26,
                 from C:\Users\Admin\Downloads\minigotchi-ESP32\config.h:26,
                 from C:\Users\Admin\Downloads\minigotchi-ESP32\minigotchi-ESP32.ino:23:
C:\Users\Admin\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.1-b6b4727c58\esp32c6/include/esp_wifi/include/esp_private/wifi_os_adapter.h:84:29: error: operator '||' has no left operand
   84 | #if CONFIG_IDF_TARGET_ESP32 || CONFIG_IDF_TARGET_ESP32S2
      |                             ^~
C:\Users\Admin\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.1-b6b4727c58\esp32c6/include/esp_wifi/include/esp_private/wifi_os_adapter.h:116:29: error: operator '!' has no right operand
  116 | #if !CONFIG_IDF_TARGET_ESP32
      |                             ^
exit status 1

Compilation error: exit status 1
```

I have the flags in the correct files

dj1ch, 2024-08-19:
Haven't tried the 3.0.x method myself, it was added by someone else. If the board is supported by 2.0.10 I would use that. I'll look into this ASAP.

bmorcelli, 2024-08-19:
What board are you compiling? Is it the esp32c6?

I ask to try myself here and find a way to solve it

closed 2024-11-28 by github-actions[bot]