dj1ch/minigotchi-ESP32 #86: display issue

cinouw, 2024-08-21:
hi i am using a lilygo t display s3 but no matter what I've tried i couldn't get the display to start working can you please help

I configured the config file correctly but event after that it didn't work

[minigotchi-ESP32.zip](https://github.com/user-attachments/files/16692585/minigotchi-ESP32.zip)

[config.zip](https://github.com/user-attachments/files/16692593/config.zip)

dj1ch, 2024-08-22:
Did you follow the install guide for the TFT libraries for the T Display S3? It is linked in the install guide

https://github.com/Xinyuan-LilyGO/T-Display-S3/tree/main?tab=readme-ov-file#4%EF%B8%8F%E2%83%A3--arduino-ide-manual-installation

cinouw, 2024-08-22:
> Did you follow the install guide for the TFT libraries for the T Display S3? It is linked in the install guide
> 
> https://github.com/Xinyuan-LilyGO/T-Display-S3/tree/main?tab=readme-ov-file#4%EF%B8%8F%E2%83%A3--arduino-ide-manual-installation

Update : the issue is fixed

So I actually did follow the guide but it didn't work either I needed to enable erase all of the flash before flashing option to get it fixed

closed 2024-08-22 by dj1ch