dj1ch/minigotchi-ESP32 #101: compilation error

LenoGhG, 2025-02-02:
i wanted to upload it o my esp32 dev board but it didnt work and i got this error message

c:/users/revi/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: C:\Users\revi\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.10/tools/sdk/esp32/lib\libnet80211.a(ieee80211_output.o): in function `ieee80211_raw_frame_sanity_check':
(.text.ieee80211_raw_frame_sanity_check+0x5c): multiple definition of `ieee80211_raw_frame_sanity_check'; C:\Users\revi\AppData\Local\arduino\sketches\32E065ED0ADDAE4BEFA02F90D237ABFD\sketch\deauth.cpp.o:C:\Users\revi\Downloads\minigotchi-ESP32-v3.5.2-beta\minigotchi-ESP32/deauth.cpp:36: first defined here
collect2.exe: error: ld returned 1 exit status

exit status 1

Compilation error: exit status 1

my name is not revi

dj1ch, 2025-02-03:
Hello, this issue can be solved with the `platform.txt` patch you need to apply as mentioned in this [section of the install guide](https://github.com/dj1ch/minigotchi-ESP32/blob/main/INSTALL.md#step-2-building-and-flashing). Let me know if you need any additional help!

LenoGhG, 2025-02-03:
okay let me try

LenoGhG, 2025-02-03:
bruh its to big i used 16mb and 80mhz flash and flash mode QIO

dj1ch, 2025-02-03:
in `config.h`, have you set `disp` to `0`? That should reduce the amount of screen libraries. Usually you can change the board to `ESP32 Dev Module` then change the settings, partitioning, etc to your liking and the sketch should fit.

LenoGhG, 2025-02-03:
Okay lemme try

LenoGhG, 2025-02-03:
I made it disp 0 and still same error message can u give me a bin file for disp 0

dj1ch, 2025-02-03:
the multiple definitions error or the not enough space error? remember that the partitioning must be `minimal spiffs` or whatever it was called.

LenoGhG, 2025-02-04:
Okay send it later when u at home

dj1ch, 2025-02-04:
i'm probably not gonna be able to build one for you, you'll need to build this like everyone else. the fix itself is just a change to the arduino-esp32 version, then modifying the platform.txt for it to run. unless it is a huge problem that cannot be solved i'll provide one but i've dealt with this issue hundreds of times so i can say that this is a pretty simple fix.

LenoGhG, 2025-02-04:
Okay thx for the help

LenoGhG, 2025-02-15:
i forgot that im pretty dumb at coding

dj1ch, 2025-02-16:
Is it still giving you the same error? What's the ESP32 board's version in the IDE? If you want you could just copy paste [this](https://github.com/dj1ch/minigotchi-ESP32/blob/main/misc/2.0.10/platform.txt) for the `platform.txt` for version `2.0.10`

closed 2025-03-03 by github-actions[bot]