dj1ch/minigotchi-ESP32 #102: Error wihle ''build''

Lyxc-Li, 2025-02-23:
Here is the error:
c:/users/user/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: C:\Users\USER\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.10/tools/sdk/esp32/lib\libnet80211.a(ieee80211_output.o): in function `ieee80211_raw_frame_sanity_check':
(.text.ieee80211_raw_frame_sanity_check+0x5c): multiple definition of `ieee80211_raw_frame_sanity_check'; C:\Users\USER\AppData\Local\arduino\sketches\854C354BBDACA4FB390AF7EF0538EE3D\sketch\deauth.cpp.o:C:\Users\USER\Desktop\minigotchi-ESP32-v3.5.2-beta\minigotchi-ESP32/deauth.cpp:36: first defined here
collect2.exe: error: ld returned 1 exit status

exit status 1

Compilation error: exit status 1
some info

[image]

[image]

dj1ch, 2025-02-23:
Pretty much the same as #101, make sure you're following the install guide and using the correct esp-idf version and it's platform.txt. Let me know if you need anything in case something weird arises.

Lyxc-Li, 2025-02-23:
> Pretty much the same as [#101](https://github.com/dj1ch/minigotchi-ESP32/issues/101), make sure you're following the install guide and using the correct esp-idf version and it's platform.txt. Let me know if you need anything in case something weird arises.

Got it to work but this is diabolycal:

[image]

closed 2025-02-23 by Lyxc-Li

dj1ch, 2025-02-23:
switch your partitioning to minimal spiffs, that might fix it.

Lyxc-Li, 2025-02-23:
woked but can am I just fing stuip or why is no acces point getting created?

Lyxc-Li, 2025-02-23:
what is the passwod and where can I coustomize it? for the accespoint propaly stands somewhere but I cant find it

dj1ch, 2025-02-23:
password is `dj1ch-minigotchi`, sorry i haven't been able to add that to the install guide