dj1ch/minigotchi-ESP32 #106: Add support for ESPBOY

coppaxamsukk, 2025-03-12:
Espboy is a diy esp8266 based project that's similar to cardputer but available from long ago. 

Can you port the awesome project for espboy?
https://www.espboy.com/

[image]

dj1ch, 2025-03-13:
Hi, if you're not aware, the Minigotchi was actually originally meant for the ESP8266. The code here is kind of just a more updated version for the newer ESP32-* ICs from Espressif.

As for the screen, I'd need to do more research. You can get it running easily by flashing it with the [ESP8266 firmware](https://github.com/dj1ch/minigotchi) but the screen obviously won't work, it seems to be an `ST7735` while you can only really use smaller screens like the `SSD1306`, `SSD1305`, `SH1106`, etc with the ESP8266 Minigotchi.

My time is rather limited here so I might not be able to do this. If you want to ask some other developer to help you that's fine by me.

coppaxamsukk, 2025-03-13:
Yeah, i know minigotchi was for esp8266, i just mistakenly added issue here.
And i mean to want port for screen actually. 
Espboy is awesome and already has (GSM phone
GPS navigator
Meteostation
LORA messenger
RFID/NFC reader/repeater
Sub1Ghz inspector
Sub1Ghz scanner
MOD/S3M/XM/IT player
AY38910/12 Yamaha chip player
Web radio
FM radio
Walkie-talkie
Meccano rover
Flower watering
Ice fishing automation
Volt/Amper meter
CW radio beacon
RSS reader
Digital ruler
Pinger
Digital thermometer
Heart rate meter
WiFi analizer
WiFi sniffer
WiFi packet monitor
WiFi pinger
WiFi deauther
IR blaster
Geiger counter
I2C scanner
MP3 player)
So i thought minigotchi would be awesome ❤️

closed 2025-03-28 by github-actions[bot]