dj1ch/minigotchi-ESP32 #107: Stuck on Channel 1

pfefferle, 2025-04-13:
**Describe the bug**
I run a minigotchi on an M5 Atom Lite and it seems to be stuck in Channel 1. At least I assume that it should switch channels!?!

Here is a log:

```
(^-^) RSSI: -47
(^-^) Channel: 5
(^-^) BSSID: XX:XX:XX:XX:XX:XX
(^-^) ESSID: {"pal":true,"name":"Palnagotchi","face":"(>__<)","epoch":1,"grid_version":"1.10.3","identity":"<removed>","pwnd_run":0,"pwnd_tot":0,"session_id":"XX:XX:XX:XX:XX:XX","timestamp":0,"uptime":0,"version":"1.8.4","policy":{"advertise":true,"bond_encounters_factor":20000,"bored_num_epochs":0,"sad_num_epochs":0,"excited_num_epochs":9999}}
 
(^-^) Successfully parsed json!
 
(^-^) Palnagotchi name: Palnagotchi
(^-^) Pwned Networks: 0
 (;-;) No Pwnagotchi found
 
(>-<) Starting advertisment...
 
(>-<) Packets per second: 1.65 pkt/s (Channel: 1)
(>-<) Packets per second: 2.81 pkt/s (Channel: 1)
(>-<) Packets per second: 3.68 pkt/s (Channel: 1)
(>-<) Packets per second: 4.34 pkt/s (Channel: 1)
(>-<) Packets per second: 4.87 pkt/s (Channel: 1)
(>-<) Packets per second: 5.31 pkt/s (Channel: 1)
(>-<) Packets per second: 5.66 pkt/s (Channel: 1)
(>-<) Packets per second: 5.97 pkt/s (Channel: 1)
(>-<) Packets per second: 6.22 pkt/s (Channel: 1)
(>-<) Packets per second: 6.45 pkt/s (Channel: 1)
(>-<) Packets per second: 6.64 pkt/s (Channel: 1)
(>-<) Packets per second: 6.81 pkt/s (Channel: 1)
(>-<) Packets per second: 6.97 pkt/s (Channel: 1)
(>-<) Packets per second: 7.10 pkt/s (Channel: 1)
(>-<) Packets per second: 7.23 pkt/s (Channel: 1)
(>-<) Packets per second: 7.34 pkt/s (Channel: 1)
(>-<) Packets per second: 7.44 pkt/s (Channel: 1)
(>-<) Packets per second: 7.53 pkt/s (Channel: 1)
(>-<) Packets per second: 7.61 pkt/s (Channel: 1)
(>-<) Packets per second: 7.69 pkt/s (Channel: 1)
(>-<) Packets per second: 7.76 pkt/s (Channel: 1)
(>-<) Packets per second: 7.83 pkt/s (Channel: 1)
(>-<) Packets per second: 7.89 pkt/s (Channel: 1)
(>-<) Packets per second: 7.94 pkt/s (Channel: 1)
(>-<) Packets per second: 8.00 pkt/s (Channel: 1)
(>-<) Packets per second: 8.05 pkt/s (Channel: 1)
(>-<) Packets per second: 8.09 pkt/s (Channel: 1)
(>-<) Packets per second: 8.14 pkt/s (Channel: 1)
(>-<) Packets per second: 8.18 pkt/s (Channel: 1)
(>-<) Packets per second: 8.22 pkt/s (Channel: 1)
(>-<) Packets per second: 8.25 pkt/s (Channel: 1)
(>-<) Packets per second: 8.29 pkt/s (Channel: 1)
(>-<) Packets per second: 8.32 pkt/s (Channel: 1)
(>-<) Packets per second: 8.35 pkt/s (Channel: 1)
(>-<) Packets per second: 8.38 pkt/s (Channel: 1)
(>-<) Packets per second: 8.41 pkt/s (Channel: 1)
(>-<) Packets per second: 8.44 pkt/s (Channel: 1)
(>-<) Packets per second: 8.46 pkt/s (Channel: 1)
(>-<) Packets per second: 8.49 pkt/s (Channel: 1)
(>-<) Packets per second: 8.51 pkt/s (Channel: 1)
(>-<) Packets per second: 8.53 pkt/s (Channel: 1)
(>-<) Packets per second: 8.55 pkt/s (Channel: 1)
(>-<) Packets per second: 8.57 pkt/s (Channel: 1)
(>-<) Packets per second: 8.59 pkt/s (Channel: 1)
(>-<) Packets per second: 8.61 pkt/s (Channel: 1)
(>-<) Packets per second: 8.63 pkt/s (Channel: 1)
(>-<) Packets per second: 8.65 pkt/s (Channel: 1)
(>-<) Packets per second: 8.66 pkt/s (Channel: 1)
(>-<) Packets per second: 8.68 pkt/s (Channel: 1)
(>-<) Packets per second: 8.69 pkt/s (Channel: 1)
(>-<) Packets per second: 8.71 pkt/s (Channel: 1)
(>-<) Packets per second: 8.72 pkt/s (Channel: 1)
(>-<) Packets per second: 8.74 pkt/s (Channel: 1)
```

**To Reproduce**
Steps to reproduce the behavior:
1. Flash Atom Lite
2. Check serial lof

**Expected behavior**
I thought it should change channels.

**Screenshots**
If applicable, add screenshots to help explain your problem.

dj1ch, 2025-04-13:
Will look into this later... thanks! 😅

pfefferle, 2025-04-14:
Thanks a lot!

pfefferle, 2025-04-22:
😳

closed 2025-05-21 by github-actions[bot]